Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN #5: How to set upload interval?

knutek66, 2021-04-13:
The Lorawan works, but interval for upload is 30 sec. How to set to 20 min?
And is the module in sleep mode to save battery?
I tried set TX_INTERVAL to 900 (15 min) but got no update at all.

knutek66, 2021-04-16:
I installed a 18650 battery 2400mAh. Still only last one night. If Lorewan dont get any battery optimiced, its not an alternative to wifi.

knutek66, 2021-04-21:
Still no one that can help. Is there any other prodjects using the TTGO HIgrow with Lorawan? I need a Higrow with TTN integration and Cayenne integration. It is working with this github code: 
[image]
but the battery drains out to fast and i dont get utdates to be every 30 min.

LorenzoComparini, 2021-06-04:
@knutek66 did you solved this? Because i would like too to set an higher upload interval, and maybe put a deepsleep mode for the ESP32

ro85ac, 2022-01-12:
@knutek66 

You need to add these pieces of code:

in setup method : `esp_sleep_enable_timer_wakeup(900* 1000000 );`
in the final method after all readings are done and the message is sent to lora, you need to call the following method:
```
Serial.println("Going to sleep now");
Serial.flush(); 
esp_deep_sleep_start();
```

If you cannot handle, put your code here (only the main class) and I will reply you!

knutek66, 2022-03-16:
Thank you for response. 
Can you show me where to add the code?

knutek66, 2022-03-16:
Here is my code. Dev id is hided.

/*
This example shows how to use LilyGO T-HiGrow- Lora SHIELD to connect to TTN and display the data in Cayenne.
2019/12/31 lewis he
*/

#include <BH1750.h>
#include <Adafruit_BME280.h>
#include <OneButton.h>
#include <DHT12.h>
#include <CayenneLPP.h>
#include "lmic.h"
#include <hal/hal.h>

#define I2C_SDA             25
#define I2C_SCL             26
#define DHT12_PIN           16
#define BAT_ADC             33
#define SALT_PIN            34
#define SOIL_PIN            32
#define BOOT_PIN            0
#define POWER_CTRL          4
#define USER_BUTTON         35
#define DS18B20_PIN         21                  //18b20 data pin

#define RADIO_RESET         23
#define RADIO_DIO_0         14
#define RADIO_DIO_1         13
#define RADIO_DIO_2         15
#define RADIO_CS            18
#define RADIO_MISO          19
#define RADIO_MOSI          27
#define RADIO_CLK           5

// Chose LSB mode on the console and then copy it here.
static const u1_t PROGMEM APPEUI[8] = { HIDING};
// LSB mode
static const u1_t PROGMEM DEVEUI[8] = { HIDING };
// MSB mode
static const u1_t PROGMEM APPKEY[16] = { HIDING };

BH1750 lightMeter(0x23); //0x23
Adafruit_BME280 bme;     //0x77
DHT12 dht12((int)DHT12_PIN, true);
OneButton button(USER_BUTTON, true);
CayenneLPP lpp(160);

static osjob_t sendjob;
String lora_msg = "";
int spreadFactor = DR_SF7;
int joinStatus = EV_JOINING;

bool lightSensorDetected = true;
bool bmeSensorDetected = true;
bool buttonClick = false;

const unsigned TX_INTERVAL = 100;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = RADIO_CS,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = RADIO_RESET,
    .dio = {RADIO_DIO_0, RADIO_DIO_1, RADIO_DIO_2}  // Pins for the Heltec ESP32 Lora board/ TTGO Lora32 with 3D metal antenna
};

void os_getArtEui (u1_t *buf)
{
    memcpy_P(buf, APPEUI, 8);
[... 219 lines cut ...]

    if (!bme.begin()) {
        Serial.println(F("Could not find a valid BME280 sensor, check wiring!"));
        bmeSensorDetected = false;
    }

    if (!lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE)) {
        Serial.println(F("Error initialising BH1750"));
        lightSensorDetected = false;
    }

    loarSetup();

    button.attachLongPressStart([] {
        // no use
    });
    button.attachLongPressStop([] {
        // no use
    });
}

void loop()
{
    os_runloop_once();
}

knutek66, 2022-03-30:
figured out how to set the code.
Now its working with 15 min intervall!
Thank you.
And with webhook it works in The things network v3

reivis, 2022-04-11:
> figured out how to set the code. Now its working with 15 min intervall! Thank you. And with webhook it works in The things network v3

Hi Knutek66, can you tell me where to put the code please.

knutek66, 2022-04-11:
put in new code in void onEvent and void setup New code marked:
void onEvent (ev_t ev)
{
    Serial.print(os_getTime());
    Serial.print(": ");
    switch (ev) {
    case EV_TXCOMPLETE:
        Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));

        if (LMIC.txrxFlags & TXRX_ACK) {
            Serial.println(F("Received ack"));
            lora_msg =  "Received ACK.";
        }

        lora_msg = "rssi:" + String(LMIC.rssi) + " snr: " + String(LMIC.snr);

        if (LMIC.dataLen) {
            // char TTN_response[256];
            // int i = 0;
            // data received in rx slot after tx
            Serial.print(F("Data Received: "));
            // Serial.write(LMIC.frame + LMIC.dataBeg, LMIC.dataLen);
            // Serial.println();
            // Serial.println(LMIC.rssi);
        }
        // Schedule next transmission
        os_setTimedCallback(&sendjob, os_getTime() + sec2osticks(TX_INTERVAL), do_send);
        
        // new code put in to set sleep mode (2022)
        Serial.println("Going to sleep now");
        Serial.flush(); 
        esp_deep_sleep_start();**
        // end of new code
        
        break;
    case EV_JOINING:
        Serial.println(F("EV_JOINING: -> Joining..."));
        lora_msg = "OTAA joining....";
        joinStatus = EV_JOINING;
        break;
    case EV_JOIN_FAILED:
        Serial.println(F("EV_JOIN_FAILED: -> Joining failed"));
        lora_msg = "OTAA Joining failed";
        break;
    case EV_JOINED:
        Serial.println(F("EV_JOINED"));
        lora_msg = "Joined!";
        joinStatus = EV_JOINED;
        delay(3);
        // Disable link check validation (automatically enabled
        // during join, but not supported by TTN at this time).
        LMIC_setLinkCheckMode(0);

        break;
    case EV_RXCOMPLETE:
        // data received in ping slot
        Serial.println(F("EV_RXCOMPLETE"));
        break;
    case EV_LINK_DEAD:
        Serial.println(F("EV_LINK_DEAD"));
        break;
    case EV_LINK_ALIVE:
        Serial.println(F("EV_LINK_ALIVE"));
        break;
    default:
        Serial.println(F("Unknown event"));
        break;
    }
}

void setup()
{
    Serial.begin(115200);

    Wire.begin(I2C_SDA, I2C_SCL);
    //new code 2022. 1190 is sleep time in seconds making 20 min sleep
    esp_sleep_enable_timer_wakeup(1190* 1000000 );
    // end new code
    //! Sensor power control pin , use deteced must set high
    pinMode(POWER_CTRL, OUTPUT);
    digitalWrite(POWER_CTRL, HIGH);
    delay(200);

    dht12.begin();

    if (!bme.begin()) {
        Serial.println(F("Could not find a valid BME280 sensor, check wiring!"));
        bmeSensorDetected = false;
    }

    if (!lightMeter.begin(BH1750::CONTINUOUS_HIGH_RES_MODE)) {
        Serial.println(F("Error initialising BH1750"));
        lightSensorDetected = false;
    }

    loarSetup();

    button.attachLongPressStart([] {
        // no use
    });
    button.attachLongPressStop([] {
        // no use
    });
}

edegraaff, 2022-12-08:
Does someone have good readings for the nutrient values?  imho they are wrong with no soil dry electrode i get a reading of 20% and when there is nothing i expect nothing :)

edegraaff, 2022-12-08:
> Still no one that can help. Is there any other prodjects using the TTGO HIgrow with Lorawan? I need a Higrow with TTN integration and Cayenne integration. It is working with this github code: [image] but the battery drains out to fast and i dont get utdates to be every 30 min.

hajhaj, 

The value for the salt/nutrient value in your example is also low = 0 seems strange right?

edegraaff, 2022-12-08:
> 

Could we link?  i want to get it working within the things network, curious how you managed to get this working
eelco nl13974