Xinyuan-LilyGO/LilyGO-T-HiGrow-LoRaWLAN #6: Lora Invalid Pins

ro85ac, 2022-01-12:
I am trying to send a Lora message using this code and pins configuration from the current github project, but it's not working:

> Setup Lora freq : 868000000
> LORA Begin FAIL
> 

The same code I use with SIM7000g board + Lora shield and it's working (of course different pins assignment).

I suppose that the pins are mapped incorrectly. If we look into the [shield description](https://ae01.alicdn.com/kf/H19c6afb44912457d8cda1f2e4e267aeat.jpg) is says that the DIO is mapped to pin 26, but pin 26 doesn't exist.

Please confirm the pins, otherwise the lora higrow shield was damaged during shipping and I will order a new one.

```
#include <SPI.h>
#include <LoRa.h>

#define RADIO_RESET         23
#define RADIO_DIO_0         14
 #define RADIO_DIO_1         13
#define RADIO_DIO_2         15
#define RADIO_CS            18
#define RADIO_MISO          19
#define RADIO_MOSI          27
#define RADIO_CLK           5

#define BAND                868E6

uint32_t counter = 0;

void setup() {
    Serial.begin(115200);
    Serial.println("Starting...");
    while (!Serial);

    Serial.println("LoRa Sender");
    SPI.begin(RADIO_CLK, RADIO_MISO, RADIO_MOSI, RADIO_CS);

    LoRa.setSPI(SPI);
    delay(1500);
    
    LoRa.setPins(RADIO_CS, RADIO_RESET, RADIO_DIO_0);

    Serial.printf("Setup Lora freq : %.0f\n", BAND);
    if (!LoRa.begin(BAND)) {
        Serial.println("LORA Begin FAIL");
        while (1);
    }
    Serial.println("LORA Begin PASS");
}
 
void loop()
{
    Serial.print("Sending packet: ");
    Serial.println(counter);

    // send packet
    LoRa.beginPacket();
    LoRa.print("hello ");
    LoRa.print(counter);
    LoRa.endPacket();

    counter++;

    delay(5000);
}
```

ro85ac, 2022-01-21:
Also the higrow board is making a strage noise and warms up when connected to lora shield

https://user-images.githubusercontent.com/11976725/150478735-a56298cf-df66-4996-b403-cb77d8eeb714.MOV

liangyingy, 2022-01-25:
> 

LilyGO T-HiGrow and T-HiGrow Lora Sheld may touch together, resulting in a short circuit.

[image]

liangyingy, 2022-01-26:
@ro85ac 

1. Does t-Higrow work on its own?

2. I see a foreign body on the T-Higrow and suggest cleaning it.

3. Please line the pin base on the T-higrow weld and then attach the.The LORa Shield SMD element may have come into contact with the baseboard component.

ro85ac, 2022-01-30:
@liangyingy the problem was soldering the lora shield to the higrow board.
Now the communication is working. The problem left is the strange noise.[image]

closed 2022-01-30 by ro85ac